scpcom/sophgo-sg200x-debian #1: [Improvement] Can you please improve documentation slightly?

Joshndroid, 2025-03-11:
Hey @scpcom 
Awesome to see someone working on a ubuntu/debian version.
I understand your likely under the pump testing builds etc but as an outsider I am unsure on what I need to install to get the best chance of being able to provide some decent feedback.
I was wondering if you could just update your documentation slightly so that we can get an idea of a couple of things.

I am using your https://scpcom.github.io as a basis for my thoughts.

1. Installation order?
Within you documentation it lists adding a debian repo. Usually I would do this once I have flashed and booted a device but it's not clear. It appears you haven't mentioned the flashing of your storage medium at all within the documentation. This can be a simple "Once you have flashed your MicroSD card with the images obtained from above link and boot the device with an internet connection, add the debian repo using the following commands." 

2. Required packages?
I understand how to flash storage mediums, and utilise a linux distro but within your list of extended packages it doesn't make clear which if/any are required for the device to even function. Can we get a 'required packages' and/or a super brief mention of what the package does so we can know if we need it for the device to function.

3. What's working/broken?
Can we get a super brief list of what is working/broken.
I would like to be able to flash this and try and help where I can, but if the image is still super early with a lot of things broken I may just be wasting my and your time noting things or issues that you are fully aware are broken/buggy.

Thanks once again I have been waiting for something like that to dust these off.

p-f-w, 2025-03-11:
hello,

i tried the installation with "licheervnano-kvm_sd.img.lz4" two times and the server will every times fail to start or crash, ssh login ist possible, no oled-display ist detected

have i selected the wrong image for -->
"Sipeed NanoKVM IP-KVM Mini Remote Control Operations Maintenance Server, 1TOPS NPU 1GHz C906 RISC-V Linux Single Board Computer, mit HDMI 100M Netzwerkanschluss (NanoKVM Full Set)"
https://amzn.eu/d/ge3EteT

?

with original 32GB sdcard the display is functional, login via usb-rndis and ssh is also working

Joshndroid, 2025-03-11:
> hello,
> 
> i tried the installation with "licheervnano-kvm_sd.img.lz4 v1.5.25" on an 256GB sdcard two times and the server will every times fail to start or crash, ssh login ist possible, no oled-display is detected
> 
> have i selected the wrong image for --> "Sipeed NanoKVM IP-KVM Mini Remote Control Operations Maintenance Server, 1TOPS NPU 1GHz C906 RISC-V Linux Single Board Computer, mit HDMI 100M Netzwerkanschluss (NanoKVM Full Set)" https://amzn.eu/d/ge3EteT
> 
> ?
> 
> with original 32GB sdcard the display is functional, login via usb-rndis and ssh is also working

Hello,
I have the same nanoKVM and just did my first install.
I used the same image as yourself.
On windows you need to get the lz4 and run the cmd command to decompress the lz4 into your img file.
https://github.com/lz4/lz4/releases
 `lz4 -d file.lz4 outputfile`
Then I used rufus to flash the image to the sdcard
The device takes a while to boot up however i have the LED screen working on the device
Within the about device section it lists the image as - licheervnano-kvm_sd-v1.5.26.img

scpcom, 2025-03-12:
Yes, sorry the v1.5.25 image had a bug while loading the kernel modules which made the NanoKVM-Server fail to start.
And this is fixed in v1.5.26.

PS: If you do not want to write the SD card again please wait for the v1.5.27, then you can fix the existing image with:
```
sudo apt-get update
sudo apt-get dist-upgrade
```

p-f-w, 2025-03-12:
thanks,

with upgrade to v1.5.27 the display works

(but the screen on loaded website in browser further only black, but today morning i had no time to check the logs)

Joshndroid, 2025-03-12:
> thanks,
> 
> with upgrade to v1.5.27 the display works
> 
> (but the screen on loaded website in browser further only black, but today morning i had no time to check the logs)

Check that webrtc is enabled within your browser. I recieved blank black screen when this was not enabled.