scpcom/sophgo-sg200x-debian #2: [Updating] Change update links within nanokvm interface 'check for updates' section

Joshndroid, 2025-03-11:
I was unsure if the updates within the original nanokvm section was for firmware outside of the main install.
I attempted it with full knowledge it may soft brick the device.
This occurred.
Is there a method to update the download links for updating to your github links so these can be installed from the device rather than needing to manually flash updates each time?

Joshndroid, 2025-03-12:
I see you sort of reference this within the issue #1 with the ability to update via the command line.
This issue still stands, or possibly just remove the ability to see the 'check for updates within the interface?'

scpcom, 2025-03-13:
I changed the update url, now it uses my github site for the update check and the zip download which contain the same files like in the debian package.
For existing images you have to use apt-get one time again to get the files with the replaced urls.
Images with v1.5.28 and later already contain the new update url.
Later I may change the web updater to use apt-get directly.

closed 2025-03-13 by scpcom