scpcom/sophgo-sg200x-debian #6: [Error] Unable to Update via terminal -

Joshndroid, 2025-03-24:
Attempted to run an update today and see that I have an issue preventing it.

The deb doesn't want to unpack....

```
Preparing to unpack .../linux-image-5.10.230-20241229-6+licheervnano-kvm_5.10.230-20241229-6+licheervnano-kvm-1_riscv64.deb ...
Unpacking linux-image-5.10.230-20241229-6+licheervnano-kvm (5.10.230-20241229-6+licheervnano-kvm-1) over (5.10.230-20241229-6+licheervnano-kvm-1) ...
dpkg: error processing archive /var/cache/apt/archives/linux-image-5.10.230-20241229-6+licheervnano-kvm_5.10.230-20241229-6+licheervnano-kvm-1_riscv64.deb (--unpack):
 unable to make backup link of './boot/System.map-5.10.230-20241229-6+licheervnano-kvm' before installing new version: Operation not permitted
dpkg-deb: error: paste subprocess was killed by signal (Broken pipe)
I: u-boot-menu: syncing /usr/lib/linux-image-5.10.230-20241229-6+licheervnano-kvm to /boot/fdt/linux-image-5.10.230-20241229-6+licheervnano-kvm
P: Checking for EXTLINUX directory... found.
P: Writing config for vmlinuz-5.10.230-20241229-6+licheervnano-kvm...
P: Writing config for vmlinuz-5.10.230-20241229-6+licheervnano...
Errors were encountered while processing:
 /var/cache/apt/archives/linux-image-5.10.230-20241229-6+licheervnano-kvm_5.10.230-20241229-6+licheervnano-kvm-1_riscv64.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

I tried; 
sudo apt clean, 
sudo apt autoremove, 
sudo apt install -f, 
sudo dpkg -i --force-overwrite /var/cache/apt/archives/linux-image-5.10.230-20241229-6+licheervnano-kvm_5.10.230-20241229-6+licheervnano-kvm-1_riscv64.deb,

None of this wants to work.
Is there something I can further try or is this going to require a reflash?

scpcom, 2025-03-25:
Yes, I am already try to find a permanent solution.
As a workaround you can do this:
```
sudo rm /boot/*5.10.230-20241229-6*
sudo apt-get install
sudo apt-mark hold linux-image-5.10.230-20241229-6+licheervnano-kvm
```

Joshndroid, 2025-03-25:
Thanks mate.
I did Google and also check the changelog before making the issue.
I'll close it

closed 2025-03-25 by Joshndroid