scpcom/sophgo-sg200x-debian #10: dpkg killed due to OOM

howels, 2025-04-22:
I tried stopping the `nanokvm` and `hostapd` services during the upgrade but dpkg cannot complete the kernel update due to memory shortage:
```
# apt upgrade
The following package was automatically installed and is no longer required:
  libicu72
Use 'sudo apt autoremove' to remove it.

Upgrading:
  board-support-licheervnano-kvm  gpg        libgpg-error0  libsensors-config                                 linux-image-licheervnano-kvm  usb-device-licheervnano-kvm
  dirmngr                         gpg-agent  libi2c0        libsensors5                                       linux-libc-dev                usbutils
  ethtool                         gpgconf    libkeyutils1   libunistring5                                     nanokvm-licheervnano          wifi-builtin-licheervnano-kvm
  gnupg                           gpgsm      libnpth0t64    libxml2                                           openssl                       wpasupplicant
  gnupg-l10n                      i2c-tools  libpcsclite1   linux-image-5.10.235-20250403-6+licheervnano-kvm  python3-requests

Installing dependencies:
  overlayfs-tools  zram-config-licheervnano

Summary:
  Upgrading: 29, Installing: 2, Removing: 0, Not Upgrading: 0
  1 not fully installed or removed.
  Download size: 0 B / 27.5 MB
  Space needed: 52.8 MB / 5872 MB available

Continue? [Y/n] y
perl: warning: Setting locale failed.
perl: warning: Please check that your locale settings:
	LANGUAGE = (unset),
	LC_ALL = (unset),
	LC_CTYPE = (unset),
	LC_NUMERIC = (unset),
	LC_COLLATE = (unset),
	LC_TIME = (unset),
	LC_MESSAGES = (unset),
	LC_MONETARY = (unset),
	LC_ADDRESS = (unset),
	LC_IDENTIFICATION = (unset),
	LC_MEASUREMENT = (unset),
	LC_PAPER = (unset),
	LC_TELEPHONE = (unset),
	LC_NAME = (unset),
	LANG = "en_GB.UTF-8"
    are supported and installed on your system.
perl: warning: Falling back to the standard locale ("C").
locale: Cannot set LC_CTYPE to default locale: No such file or directory
locale: Cannot set LC_MESSAGES to default locale: No such file or directory
locale: Cannot set LC_ALL to default locale: No such file or directory
Extracting templates from packages: 100%
(Reading database ... 42065 files and directories currently installed.)
Preparing to unpack .../00-linux-image-5.10.235-20250403-6+licheervnano-kvm_5.10.235-20250403-6+licheervnano-kvm-1_riscv64.deb ...
Unpacking linux-image-5.10.235-20250403-6+licheervnano-kvm (5.10.235-20250403-6+licheervnano-kvm-1) over (5.10.235-20250403-6+licheervnano-kvm-1) ...
Killed
```

Any suggestions?

scpcom, 2025-04-22:
Did you stop nanokvm with this command?
`sudo systemctl stop nanokvm`
After this you can continue the update with:
`sudo apt-get install`

PS: The update will also install zram-config-licheervnano, which is the solution for future updates. It creates a zswap device which prevents from OOM during apt upgrade.

howels, 2025-04-22:
> Did you stop nanokvm with this command? `sudo systemctl stop nanokvm` After this you can continue the update with: `sudo apt-get install`
>
> PS: The update will also install zram-config-licheervnano, [...]

Yes, I had stopped the `nanokvm` service. I eventually got it to complete by stopping every service I could find. This started with 1.6.3 so could have an older config.

Will all updates arrive via package updates or do we need to perform a full-image update from time to time? Presumably we can export `/etc/kvm` to capture the important aspects of the KVM app and re-import that into the new install.

Thanks for a fantastic OSS alternative to the original firmware!

scpcom, 2025-04-23:
I put almost all changes into the Debian packages, a full image update should never be required.
Maybe if Debian 13 gets officially released you could start with a fresh image (but still optional).

closed 2025-04-23 by scpcom