scpcom/sophgo-sg200x-debian #12: [Bug] dependency issue upgrading board-support-licheervnano-kvm

howels, 2025-05-24:
Trying to upgrade and hit an error in the image on one host where ca-certs seems to have not updated, which blocked an upgrade of board-support-licheervnano-kvm, then when that package updated it hit an error due to missing systemd services:

```
Preparing to unpack .../board-support-licheervnano-kvm_1.0.40_riscv64.deb ...
Unpacking board-support-licheervnano-kvm (1.0.40) ...
Setting up board-support-licheervnano-kvm (1.0.40) ...
Unit /etc/systemd/system/gadget-nic-usb0.service is added as a dependency to a non-existent unit sys-subsystem-net-devices-usb0.device.
Unit /etc/systemd/system/gadget-nic-usb1.service is added as a dependency to a non-existent unit sys-subsystem-net-devices-usb1.device.
Unit /etc/systemd/system/wifi-builtin-wlan0.service is added as a dependency to a non-existent unit sys-subsystem-net-devices-wlan0.device.
Unit /etc/systemd/system/wifi-hostapd-wlan0.service is added as a dependency to a non-existent unit sys-subsystem-net-devices-wlan0.device.
```

scpcom, 2025-05-25:
The "... is added as a dependency to a non-existent unit ..." message is only a warning, not an error (maybe I do a review on the units and see If I can remove the dependency).
If the upgrade fails with an error the issue is somewhere else.

howels, 2025-05-25:
The package fails to configure - what else can I check? I've stopped nanokvm to reclaim ram during the upgrade

scpcom, 2025-05-25:
hmm, OK, I never had SSL issues with the images. Maybe it was a failed previous upgrade as you mentioned.
What image version did you start with?
`cat /boot/ver`
Are the RAM shortages still a thing? Do you have zram-config-licheervnano installed?
If zram is running correctly it should show a 300M swap device here:
`free -m`

howels, 2025-05-25:
> hmm, OK, I never had SSL issues with the images. Maybe it was a failed previous upgrade as you mentioned. What image version did you start with? `cat /boot/ver` Are the RAM shortages still a thing? [...]

The original version is `licheervnano-kvm_sd-v1.6.3.img` before rebooting I have no zram device:

```
debian@kvm-bob:~$ free -m
               total        used        free      shared  buff/cache   available
Mem:             196          60          43           0          99         135
Swap:              0           0           0
```

but after a reboot the device has appeared:
```
debian@kvm-bob:~$ free
               total        used        free      shared  buff/cache   available
Mem:          201260       88728       13400       18408      124716      112532
Swap:         307196           0      307196
```

howels, 2025-05-25:
Closing the issue as it appears to be related to failed upgrades instead of a bug.

closed 2025-05-25 by howels