scpcom/sophgo-sg200x-debian #13: [Question] How to use and control audio using the onboard DAC?

HeinrichAD, 2025-05-26:
First, thanks for the amazing work!

I'm currently attempting to connect a speaker directly to the [Milk-V Duo S](https://milkv.io/docs/duo/getting-started/duos) (without EMMC) using the onboard DAC and the AUDIO_OUT_R and AUDIO_OUT_L outputs (Pins 49 and 51).

I was once able to play music by chosing a specific device:

```bash
aplay -D dmix:CARD=cv182xadac,DEV=0 sample-3s.wav
```

Unfortunately, I haven't been able to replicate that success. I'm also unable to control settings such as volume. As shown in the device information below, only the ADC and DAC are visible. I would expect to also see options like "Line Out" and "Line Out Source" to manage volume and switch between stereo and mono output.

Here's a comparison between what I currently see and what I would expect to see — based on the output from one of my other boards (Orange Pi Zero 3):

|Actual|Expected|
|:-:|:-:|
|[image]|[image]|

I'd also like to note that simply installing PulseAudio using `apt install pulseaudio` did not work. After installation, the service either fails to start properly or, when it does run, it provides no meaningful feedback — for example, `pacmd` with the error "Daemon not responding". I was also not successful using PipeWire.

I would greatly appreciate any assistance. How can I use the DAC with the current Debian image, and how can I control the volume freely?

<details><summary>Device Infomation</summary>

I am using the latest release `v1.6.23`.

```txt
$ uname -a
Linux duos-d298 5.10.235-20250430-6+duos #1 PREEMPT Sun May 25 13:36:19 UTC 2025 riscv64 GNU/Linux
```

```txt
$ amixer scontrols
Simple mixer control 'ADC',0
Simple mixer control 'ADC Capture Mute',0
Simple mixer control 'ADC Power',0

$ amixer -c 1 scontrols
Simple mixer control 'DAC',0
Simple mixer control 'DAC Playback MUTE',0
Simple mixer control 'DAC Playback Power Up/Down',0
```

```txt
$ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 1: cv182xadac [cv182xa_dac], device 0: cviteka-dac 300a000.dac-0 [cviteka-dac 300a000.dac-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

$ aplay -L
null
    Discard all samples (playback) or generate zero samples (capture)
hw:CARD=cv182xadac,DEV=0
    cv182xa_dac, cviteka-dac 300a000.dac-0
    Direct hardware device without any conversions
plughw:CARD=cv182xadac,DEV=0
    cv182xa_dac, cviteka-dac 300a000.dac-0
    Hardware device with all software conversions
default:CARD=cv182xadac
    cv182xa_dac, cviteka-dac 300a000.dac-0
    Default Audio Device
sysdefault:CARD=cv182xadac
    cv182xa_dac, cviteka-dac 300a000.dac-0
    Default Audio Device
dmix:CARD=cv182xadac,DEV=0
    cv182xa_dac, cviteka-dac 300a000.dac-0
    Direct sample mixing device
```

</details>

scpcom, 2025-05-26:
I think pulseaudio and PipeWire are optimized for desktop usage and may fail because there is no Desktop.
You should remove pulseaudio and PipeWire to make alsa working again.

Is this just a naming issue? ADC is Line-In an DAC is Line-Out?
I tink we need hardware-specific config files to setup the mixers (maybe alsa some config or I need to add this to the DTS of the Linux Kernel).

Can you control the volume via the "DAC" mixer?

scpcom, 2025-05-26:
I compared the drivers and found Orange Pi Zero 3 Chip has 3 different outputs, therefore it provides 3 mixers "DAC Playback Volume", "Headphone Playback Volume" and "Line Out Playback Volume".
The Milk-V DuoS Chip has only one output, "DAC Playback Volume".
alsmamixer just hides the " Playback Volume" part.

HeinrichAD, 2025-05-26:
I agree with you regarding PulseAudio and PipeWire. I only noticed this while searching for a possible solution.

> Is this just a naming issue? ADC is Line-In an DAC is Line-Out?

In my experience, that’s not the case. On my other two different boards, these are two distinct components. Additionally, the DAC tends to behave differently and is often limited when it comes to volume control. That's why I was looking for something like a "Line Out" or similar option to manage the volume.

> Can you control the volume via the "DAC" mixer?

In a way, yes ... but not entirely. When using alsamixer, no matter what I do, the volume always increases, even when pressing the down arrow. If I try setting a lower volume directly, it immediately resets to a higher level. I currently suspect that the "DAC" only accepts values between 50 and 100, and ALSA automatically maps any input value to that range. For example, setting 0 results in 50, 100 stays at 100, and something like 20 gets translated to around 60.

> I compared the drivers and found Orange Pi Zero 3 Chip has 3 different outputs, therefore it provides 3 mixers "DAC Playback Volume", "Headphone Playback Volume" and "Line Out Playback Volume".

Thanks for checking. (I should have checked that on my own :unamused:) This sure explains it. It's a bit odd that I can control the DAC volume through "Line Out", but it's not a priority right now.

I will create a new "clean" image tomorrow and test using only ALSA again. But since a value of 50 for the DAC is quite high and extremely loud even with a tiny amplifier, do you have an idea how to reduce the volume or how to gain more control over the output volume (without additional hardware)?

scpcom, 2025-05-26:
Yes, the volume range is limited by the hardware.
If I enable usb host mode and connect a USB Audio adapter to DuoS I can control the volume from 34 to 100 with 1-2 percent per step.
I do not know how to solve this yet, maybe with a software mixer.
Changes in the driver may be a better solution but I do not know enough about the DAC hardware part.

scpcom, 2025-05-27:
I found a bug and fixed it:
https://github.com/scpcom/linux/commit/a18b0983e063595b7467b7a1e35fbb9001c5b4db
Now I can control the DAC volume with full range in 2-3 percent steps.
I takes some time until the patch gets into the image and you can test it.

HeinrichAD, 2025-05-27:
**Thank you for your incredibly quick assistance and fix. Everything is now working perfectly.**

Here are a few observations I made:

- The DAC volume appears to be inverted (100 is mute, and 0 is maximum volume), though I've seen similar behavior on other devices as well.
- Audio output only works through the `dmix:CARD=cv182xadac` device, at least on my site.
- No default device is set, and since the ADC is listed before the DAC, ALSA defaults to the ADC.

I was able to resolve these issues by configuring ALSA. This can be done system-wide via `/etc/asound.conf` or per user with `~/.asoundrc`.

```rc
defaults.pcm.!card 1
defaults.ctl.!card 1

pcm.SoftVolume {
    type          softvol
    slave.pcm     "dmix:CARD=cv182xadac"
    control.name  "Master"
    control.card  1
}
pcm.!default {
    type          plug
    slave.pcm     "SoftVolume"
}
```

- The first two lines set the DAC as the default sound card.
- The `SoftVolume` block defines a new virtual device using ALSA's [softvol plugin](https://github.com/markc/alsa/blob/master/lib/md/Softvol.md). This creates a volume control interface linked to the working DAC device, but without the inverted volume behavior.
- The final block sets `SoftVolume` as the default output. With this setup, I can now play audio without specifying any device parameters, e.g., `aplay sample-3s.wav`.

Running `alsamixer` now displays the following:

[image]

Here, I set the DAC volume to zero, which effectively means 100 percent, and control the actual volume only through "Master", which is the new virtual SoftVolume device.

I hope this documentation proves helpful to the next person and saves them some time 😅.

HeinrichAD, 2025-05-27:
Here is an additional observation I just made:
- After rebooting the system the volume states are resetted. (In my case DAC to 75% and Master to 100%.)

ALSA has to options (mutually exclusive) to tackle that (as far as I know):
1. saving settings with a daemon
    - an empty file `/etc/alsa/state-daemon.conf` has to exists
    - `alsa-state` service has to be enabled and started
        ```sh
        systemctl enable alsa-state
        systemctl start alsa-state
        ```
2. saving settings at the end of sessions
    - the file `/etc/alsa/state-daemon.conf` must not exists
    - `alsa-restore` service has to be enabled and started
        ```sh
        systemctl enable alsa-restore
        systemctl start alsa-restore
        ```