scpcom/sophgo-sg200x-debian #18: Request for adding official LCD initialization in U-boot

physics1024, 2025-06-04:
Hi,

I was amazed when I first found this work and wow, that is what I really need! I've wasted a month on migrate a Debian to LicheeRV nano, however my attempt never succeeded, due to SBI version and so on. 

There is a request on enhancement: could you support the official LCD initialization? I found the Debian image could not boot my LCD  but another repository "LicheeSG nano build" could, and I noticed that there are a few differences between both U-boot. I tried to replace FIP firmware and got CGROUP v1 error, and somehow adding boot args could not save the problem. My next attempt about replacing boot commands in U-boot was banned due to my academics.

Thanks a lot about doing this great work on Debian, which greatly relieve my load on developing!

scpcom, 2025-06-05:
OK, I started the implementation yesterday. Currently I have set "zct2133v1", which is the the model of Sipeeds default fip.bin
I also enabled the video driver on u-boot.
The default panel can be changed by setting the variable PANEL_TUNING_DEFAULT.
A new image may be released this evening.
Next I need to find a way to tell the Makefile to generate multiple fip packages which would allow the user to choose the right one just by installing it via apt-get.

I did not take a change on the kernel config yet. But as far as I know there is no KMS/DRM driver, only a FB driver (FB_CVITEK).

physics1024, 2025-06-06:
Much respect to your effort! I'm now working on DirectFB2 to make sure that it will work fine when panel is ready.

scpcom, 2025-06-06:
The new image and packages are online now.

First you need to get the regular updates:
```
sudo apt-get update
sudo apt-get dist-upgrade

```

Next you can search for available fsbl (fip) packages and install one of them:
```
sudo apt-cache search cvitek-fsbl
sudo apt-get install cvitek-fsbl-licheervnano-st7701-hd228001c31

```

If you want to use the FB driver you can enable it with:
`sudo touch /boot/fb`

If everything works make sure to remove the non-LCD fsbl:
`sudo apt-get remove cvitek-fsbl-licheervnano`
And if you tried others they should be removed too, only keep the one fsbl you want.

scpcom, 2025-06-06:
The framebuffer driver worked? Did you see a text console?
I did not use the LCD before so I have not done real world testing for this component in the past.
Now I found the right model for my exotic LCD panel (d240si31) and added some extra code for it and it works so I started testing today.

For the next image release I added a tool to test the LCD panel without FB:
```
sudo rmmod soph_fb
sudo insmod /mnt/system/ko/soph_mipi_tx.ko 
sudo /mnt/system/usr/bin/sample_panel --panel=ST7701_HD228001C31 -s

```
It will be available if you install the updates via apt-get tomorrow.
Next I will test the FB

physics1024, 2025-06-07:
No, the panel didn't display anything. No console, no graphics, just whole black. My mate also tested the image and got the same situation.

I also took a look in `fbset -i`, and the result: 
```
root@licheervnano-30df:~# fbset -i

mode "320x240-1"
    # D: 0.097 MHz, H: 0.202 kHz, V: 0.564 Hz
    geometry 320 240 320 480 32
    timings 10317660 10 140 107 10 9 2
    rgba 8/0,8/8,8/16,8/24
endmode

Frame buffer device information:
    Name        : cvifb
    Address     : 0x8b980000
    Size        : 5767168
    Type        : PACKED PIXELS
    Visual      : TRUECOLOR
    XPanStep    : 1
    YPanStep    : 1
    YWrapStep   : 0
    LineLength  : 1280
    MMIO Address: 0xa088000
    MMIO Size   : 4095
    Accelerator : No
```

It looks like a driver incompatible problem, or something in FDT didn't work, because the resolution isn't correct.

scpcom, 2025-06-07:
OK, same here.
With soph_mipi_tx driver and sample_panel app the graphics (test patterns) are shown.
With soph_fb driver only black screen.
I have to test the FB a bit more.

physics1024, 2025-06-08:
In the official image the console never shows. Maybe need a terminal emulator then? DFB also provides a simple TTY emu.

scpcom, 2025-06-08:
I can enable the FB console in the kernel if required.
But if you want to use the FB for graphics output and not as terminal this step may not be required.
Before we check this, please wait for the next image/packages release and try this first. Version v1.6.31 is building now.
The updated fsbl will have bootlogo enabled and sets the correct FB parameters.

After updating you need to run this to enable the bootlogo:
`sudo mkenvimage -s 0x20000 -o /boot/uboot.env /etc/u-boot-initial-env`

scpcom, 2025-06-08:
The new packages are online now.

physics1024, 2025-06-08:
The panel successfuly run! Though FB driver is still a slow rendering method. Next I will find a way to use the 2D-Blend accelerator in sg2002 to make video output much efficient.

Thanks again to make this contribution! The issue could be closed if there are no other bugs.

scpcom, 2025-06-08:
Good to know.

Yes, FB is the slowest method, the fastest (but more complex) is direct use of the VO module from the ISP.
https://github.com/scpcom/sophgo-middleware/tree/maix_mmf-cvisdk/sample

Maybe it can be easier (or not) when using MaixCDK:
https://github.com/sipeed/MaixCDK/
I did not make a Debian package yet but you can already find the required libmaixcam_lib.so in /mnt/system/usr/lib/ which allows using MaixCDK on any sg200x hardware.
Since the official image uses musl and Debian uses glibc it can not be used directly yet. Currently I am building the MaixCDK glibc apps with my buildroot variant.
But this is a different story so I will close this issue here :-)

closed 2025-06-08 by scpcom

physics1024, 2025-12-25:
> Is this panel possible to connect with SG2000 Milk v duo s? since it offers MIPI-DSI display.

Actually DUO S seems have its own panel type, which also has a firmware in driver.