scpcom/sophgo-sg200x-debian #19: how to increase root?

rossiniscarface, 2025-07-22:
I installed successfully licheervnano-kvm_sd-v1.6.35.img on my sipeed nanokvm lite. 

debian@kvm:~$ df -h
Filesystem      Size  Used Avail Use% Mounted on
/dev/root       1.3G  1.2G   84M  94% /
devtmpfs         99M     0   99M   0% /dev
tmpfs            99M     0   99M   0% /dev/shm
tmpfs            40M  684K   39M   2% /run
tmpfs           5.0M     0  5.0M   0% /run/lock
tmpfs            99M   18M   81M  18% /tmp
/dev/mmcblk0p1  100M   11M   90M  11% /boot
/dev/zram1       53M  4.2M   44M   9% /opt/zram/zram1
overlay1         53M  4.2M   44M   9% /var/log
tmpfs            20M  4.0K   20M   1% /run/user/1000

Commands like apt upgrade are failing because not enough spce left.

closed 2025-10-19 by scpcom

rossiniscarface, 2025-11-08:
Just tried it with a fresh install. Same issue. 
I had a look into 

```
# cat /etc/systemd/system/finalize-image.service
[Unit]
Description=Finalize the Image
Before=ssh.service

[Service]
Type=oneshot
ExecStartPre=-/usr/sbin/parted -s -f /dev/mmcblk0 resizepart 2 25%
ExecStartPre=-/usr/sbin/resize2fs /dev/mmcblk0p2
ExecStartPre=-/bin/dd if=/dev/hwrng of=/dev/urandom count=1 bs=4096
ExecStartPre=-/bin/sh -c "/bin/rm -f -v /etc/ssh/ssh_host_*_key*"
ExecStart=/usr/bin/ssh-keygen -A -v
ExecStartPost=/bin/systemctl disable finalize-image
```
so it tries to resize it to 25% and not the rest of the device. 
And my device is "only" 4 GB

scpcom, 2025-11-08:
OK, 4GB is to less for the default configuration
The original Sipeed NanoKVM image is designed for 32GB cards.
My image needs 8GB card as minimum.
The other 75% are reserved for the /data partition (disk images/virtual media).

You can change the 25% to 50% to make it work.